AlgoCompSynth/CLAMS #32: Test Waveshare audio hardware

znmeb, 2023-02-24:
Hardware ordered - it looks better suited to breadboarding than the Pimoroni unit because you can connect to the UART without an extra base

znmeb, 2023-02-24:
waiting on hardware - will post ETA when I get one

znmeb, 2023-03-05:
Two Waveshare audio add-ons and two Waveshare overall eval boards are in hand. Overall board is running `zeptoforth` with a Pico H installed.

I'm debating whether to publicize the overall rig since Amazon is now out of stock and shipping from China is still painful. Still, it has a serial-to-USB (that works!), and audio jack (but no speakers) and an LCD touch screen. The documentation exists but is not great. I really don't want to "contribute" to a Chinese for-profit.

znmeb, 2023-03-08:
Moving all Waveshare hardware to the "future" project for now. I just don't have the spare hours to troubleshoot software that doesn't have an open source license or decent English language documentation.

znmeb, 2023-09-24:
Done - all that's required is knowing the DAC pin numbers

DoomHammer, 2024-01-12:
So, what was the pinout and where can we find the code?

znmeb, 2024-01-14:
> So, what was the pinout and where can we find the code?

I'm not currently working with the Waveshare devices, but the next round of code will have the I2S pins as user-specified parameters. Meanwhile, if you've got one of them, the pinout is on this page from their website:

<https://www.waveshare.com/pico-audio.htm>

DoomHammer, 2024-01-14:
Yeah, I was trying to follow this pinout using Pico Arduino Core and CircuitPython but without luck so far :( That's why I asked about the code you've used.

znmeb, 2024-01-14:
I got this demo working on the Waveshare by changing the pin numbers https://learn.adafruit.com/mp3-playback-rp2040/pico-i2s-mp3

DoomHammer, 2024-01-15:
@znmeb using the V1 board or V2.1? It seems V2.1 is a bit more problematic due to swapped LRCK and SCLK pins.

znmeb, 2024-01-15:
V2 - I also tested this board - https://www.waveshare.com/wiki/Pico-Eval-Board

DoomHammer, 2024-01-15:
So the one with Master Clock? Which version of CircuitPython did you use?

znmeb, 2024-01-15:
> So the one with Master Clock? Which version of CircuitPython did you use?

I'm not sure. Most likely it was the latest version as of mid-March 2023, which would be 8.0.4. I don't do much with CircuitPython or MicroPython, and the project was pretty much on hold from March through September, when I closed the issue.

The last I2S audio testing I did was with Pimoroni's version of MicroPython on the Pimoroni PicoVision - https://github.com/pimoroni/picovision/tree/main/modules/picosynth is the module I used. That pinout is at <https://cdn.shopify.com/s/files/1/0174/1800/files/picovision_diagram.pdf?v=1696414342>. My code is at https://github.com/AlgoCompSynth/CLAMS/blob/main/PicoVision-dev-setup/pimoroni-picovision-examples-v1.0.2/clambake.py

Sadly the PicoVision is out of stock, but Pimoroni do have a waitlist you can sign up for.

DoomHammer, 2024-01-17:
PicoVision looks neat, will see if I can grab one :)

znmeb, 2024-01-17:
The Waveshare eval board is nice - has a decent size screen, a wired UART - USB connection and an audio output. But it's more expensive than the PicoVision and doesn't do HDMI.

closed 2024-10-14 by znmeb